prusa3d/PrusaSlicer #9935: Only Infill where needed - Adaptive Cubic

kinematicdigit, 2023-03-05:
### Description of the bug

Strange behaviour here when combined with Adaptive Cubic infill and Only Infill where needed, where the infill begins midair. 

[image]

### Project file & How to reproduce

Head_For_Cape-Fixed.3mf.zip

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.0-alpha4+MacOS-arm64

### Operating system

Mac OS Ventura 13.2

### Printer model

Ender 3V2 with Sonic Pad

bubnikv, 2023-03-06:
Infill only where needed is frankly ripe for removal. I am not sure whether it is worth investing effort into fixing it.
It is long superseded with the cubic support and lightning infills.

kinematicdigit, 2023-03-06:
> Infill only where needed is frankly ripe for removal. I am not sure whether it is worth investing effort into fixing it.
> It is long superseded with the cubic support and lightning infills.

I'm pretty new to Prusa Slicer, and experiemented with the feature, but makes sense if it doesn't have value to others. I only noticed it as I was watching the printer try to drop filament in midair and then checked the slicer for this.

lukasmatena, 2023-04-02:
Fixed by removing the option in [2.6.0-alpha6](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha6). Closing.

closed 2023-04-02 by lukasmatena