prusa3d/PrusaSlicer #9937: Organic When a lot of piece

caltoyuno, 2023-03-05:
### Description of the bug

The screenshot talk for itself.
When you are a lot piece on the plate, with organic choose. 
we need to up the gap between piece to do something good.
May be in the future improse that with the branch density ? 

### Project file & How to reproduce

[2.6.0.zip](https://github.com/prusa3d/PrusaSlicer/files/10892386/2.6.0.zip)

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0 alpha 4.0

### Operating system

Windows 10

### Printer model

AnkerMake M5

kubispe1, 2023-03-08:
1) In your example, there are quite a lot of collisions between objects and supports. See #316
So if it is the point of the issue it is duplicate.
2) We would like to optimize or reduce branch density, but it will not solve 1) in general.
Can you update your issue?

caltoyuno, 2023-03-08:
Oh yes, I think the same issue that you noted, so i closed my issue :)
Sorry for that.
I hope may be this issue can be solve :)

closed 2023-03-08 by caltoyuno

bubnikv, 2023-03-09:
I suppose you don't like that the trees that are generated for different objects overlap?