prusa3d/PrusaSlicer #9938: Correct MakerGear.ini Z extra moves in start_gcode

brigthub, 2023-03-05:
### Description of the bug

The "start_gcode" variable in new MakerGear.ini alpha v0.1.0 moves the z rapidly to .3 with head over build plate, and THEN on next lines moves down to Z10 to prep for wipe.

`start_gcode = M108 S255 ; turn on M2 fans\nG28 ; home all axes\nG1 Y50 Z0.3 F9600 ; move forward to avoid binder clips\nG1 X205 Z10 ; move off platform\nG1 Z0.4 ; position nozzle\nG92 E0 ; zero extruder\nM109 S[first_layer_temperature_[current_extruder]]\nG1 E25 F225 ; purge nozzle\nG92 E0 ; zero extruder\nG1 X190 Z0.1 E1.0 F1200 ; slow wipe\nG1 X180 Z0.25 ; lift\n
`

I recommend removing the initial Z0.3 move and let the next following line perform the Z move to 10.

### Project file & How to reproduce

1. Obtain MakerGear.idx, MakeGear.ini, MakerGear directory from GIT
2. Delete PrusaSlicer directory from AppData Local directory with PriusaSlicer--not running
3. Install MakerGear files in the Programs resources profile directory
4. Run PrusaSlicer
5. Select and setup MakerGear printer
6. Slice an object
7. Print
--- result watch z movements before printing begins

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer 2.5.0+win64

### Operating system

WIndows10 22H2

### Printer model

MakerGear M2rE

rtyr, 2023-03-06:
@Garr-R what do you think?

Garr-R, 2023-03-08:
Sounds good to me! I've made the adjustment and have submitted a PR on PrusaSlicer-settings repo:
https://github.com/prusa3d/PrusaSlicer-settings/pull/187

Thank you for the suggestion! @brigthub

closed 2024-03-11 by rtyr