prusa3d/PrusaSlicer #9939: Fuzzy Skin - 'Outside Wall' setting fuzzies inside walls as well if a single perimeter outer wall exists

dstulken, 2023-03-05:
### Description of the bug

If a shape has a hollow internal volume, which is separated from the outside of the model by only a single perimeter wall, the "Outside Wall" Fuzzy Skin algorithm incorrectly also applies the fuzzy skin to all of the inside walls.  This is true even for "water tight" models with no holes or perforations.   

[image]

### Project file & How to reproduce

**How to Recreate:**
PrusaSlicer 2.5.0+MacOS-x64
Print Settings: 0.25mm DRAFT
Filament: Generic PLA
Printer: Original Prusa MINI & MINI+

Right click - add shape - box
Size: 100x100x100mm
Position: 90, 90, 50

Right click - add negative volume - box
Size: 96x96x96mm
Position: 0, 0, 0

(This creates a hollow cube with 2mm thick walls on all sides.)

Print settings - Fuzzy Skin - Outside Walls
0.3mm thickness (default)
0.8mm point distance (default)

================================

Slice
Drag layer slider down about half way
observe that fuzzy skin is correctly applied to the outside of the box
observe that fuzzy skin is correctly NOT applied to the inside of the box
observe that each side of the box is 4 perimeters thick
This is all as we would expect.  (See image below).

[image]

Now change the negative volume X position to -0.2
Slice
observe that fuzzy skin is correctly applied to the outside of the box
observe that fuzzy skin is correctly NOT applied to the inside of the box
observe that the left side of the box is now only 2 perimeters thick
This is still functioning as we would expect.  (See image below).

[image]

Now change the negative volume X position to -0.3
Slice
observe that fuzzy skin is incorrectly applied to both the inside and outside of the box
observe that the left side of the box is now only 1 perimeter thick

The "outer wall" fuzzy skin application has failed, even though our model is still fully enclosed ("water tight").
(see image below)
[image]

The relative sizes do not appear to matter - only that the outer wall is a single sliced perimeter thick.
Here is an example with the negative modifier box set to 5x5x5mm, position -8.45, 0, 0.
Note that the interior walls of the inside of this tiny internal volume are all fuzzy skinned.

[image]

============================================================

**We can also trigger this a different way:**
Reset the negative volume X position back to -0.2, 0, 0, and size back to 96x96x96mm  
(This is the offset setting that gives a 2-perimeter left side wall, which slices as expected with a fuzzy outer wall and smooth interior walls.)
Slice, and verify that this is correct as expected

Now change Printer Settings - Extruder 1 - Nozzle diameter to 0.6
Change Print Settings - Advanced - Extrusion width (all values) to zero (auto) so that the nozzle change is applied to the slicing.
Slice
observe that fuzzy skin is incorrectly applied to both the inside and outside of the box
observe that the left side of the box is only 1 perimeter thick

[image]

So the occurrence of this bug does not correspond to the absolute physical feature size (which is unchanged between these two cases, since they used the same model).

Thanks in advance, and let me know if you need any further information.

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0+MacOS-x64

### Operating system

macOS 10.14.6

### Printer model

Prusa MINI+

dstulken, 2023-03-07:
Two more data points:

1) It looks like the false classification cascades if multiple negative volumes are adjacent.  

[image]

2) Even more interesting - if two negative volumes share a single-perimeter boundary, that boundary is detected as being "outside" (and receives the fuzzy skin) even if it is completely internal to the model!
[image]

DoubleStrike, 2024-01-02:
This is also broken if you have a single modifier volume that pokes into a model. For example, a cube that has another cube that is partially sticking into the model.  It will create a second set of perimeters and fuzzy walls internally, ruining the geometry.

This bug has been open for 9 months. Any updates from the dev team?

David-tech42, 2024-02-04:
Would also love to have a fix, building modifiers all over the place is very painful workaround.

sefose, 2024-02-20:
I would also like to have a fix - especially in multicolor prints, the fuzz is at wrong places. And maybe something like paint on supports, but for fuzzy skin?

ohthetrees, 2024-03-24:
Having this problem too. 

Version 2.7.2+MacOS-arm64

[image]

closed 2026-09-29 by prusaslicer-bot[bot]