prusa3d/PrusaSlicer #9956: Additional Archive Import Format support

rcmaniac25, 2023-03-07:
**Is your feature request related to a problem? Please describe.**
The Import ZIP Archive feature of 2.6 alpha5 was the feature I didn't know I wanted. My issue now is that I get models in 2 archives: Zip (majority) and Rar (not as common), but to save space I've spent the last couple years re-archiving them as 7z. The new archive import obviously only supports zip right now.

**Describe the solution you'd like**
Support for additional archive formats like 7z.

**Describe how it would work**
Same as importing a zip archive, but additional formats.

**Describe alternatives you've considered**
I've checked miniz and to determine if it may be just drop in support for 7z, but it is not. I could simply not re-archive the existing archives, but I save anywhere from 2%-20% space with 7z over zip.

Also the obvious alt: just don't bother using the archive import. I know the existing implementation extracts to a tmp file, then reads them back in with existing implementations. So there's still a temporary file system impact when importing from archive, and simply extracting from a 7z or rar, loading, and then deleting is just a more manual way to do this.

**Additional context**
Besides 7z and rar (which is less of a request, but would be nice to have), I'm not really specifying other formats. I don't know what people use or maybe don't in general. Maybe they use tar (not really compression, but different story), gz, or tar.gz.

I've seen too many creators change between model sharing sites, or Patreons move between DropBox, MyMiniFactory, Cults, Sync, MyAirDrop, and more. to truly trust "don't bother downloading. Just leave it in the cloud". But I know there are those that do. But it means I have a lot of files downloaded in whatever format the creator or service archived them in.

I can see Printables adapting their existing Open in PrusaSlicer option to download a zip and open the file selector so you don't have to click 20 files one by one and all the browser UI that comes with it.

lukasmatena, 2023-03-07:
> I can see Printables adapting their existing Open in PrusaSlicer option to download a zip and open the file selector so you don't have to click 20 files one by one and all the browser UI that comes with it.

Isn't that exactly how it works now?

rcmaniac25, 2023-03-07:
> Isn't that exactly how it works now?

Too be honest, I haven't used it myself not tried to find the code. But I watched a video of it and it was file by file.

If anything, it's not related to the main point of the feature request and can be ignored if I am mistaken.

rcmaniac25, 2026-09-13:
3.0 alpha11 doesn't currently support ZIPs, but besides adding that back, adding more formats is still relevant.