prusa3d/PrusaSlicer #9957: Can't turn off The Newly implimented "Ensure vertical shell thickness" concentric infill in 2.6 a5

Snuff1eupagus, 2023-03-07:
### Description of the bug

There is no way to turn this off. It's nice that you have taken steps to fix the fill type and direction. However we do not always need or want this, and I can't seem to find a way to turn it off. It makes no sense to remove our control over weather we have it or  not. 

At least with "Ensure vertical shell thickness" before we could decide to turn it off or use it if needed.

I've read all the release notes and have been through the software, if I'm just missing it please let me know.

Also it appears to be putting way more material on the top slopes then on the overhanging even though they are the same angle 45 degrees.

### Project file & How to reproduce

Add a square rotate it 45 degrees Y axis, sink it slightly below the bed and slice, it doesn't matter which Perimeter gen you use.

In this example the lower layers of "Ensure vertical shell thickness" are 0.38 and the upper are approx. 0.7.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6 alpha 5

### Operating system

Win 7 - 10

### Printer model

n/a

AssiveMass, 2023-03-08:
Turning off `Ensure vertical shell thickness` never removed the solid infill on slopes, if I'm not mistaken, but switched between the old Slic3r method and the newer Prusa method. The new concentric solid infill is now the default, with no option to use the old Slic3er method.

I too wish we could turn solid infill off completely.
I'm trying to print objects with 100% infill, but solid infill is unnecessarily added here, which makes a lot of short movements on certain layers.

[image]

Snuff1eupagus, 2023-03-08:
`@AssiveMass wrote:
Turning off Ensure vertical shell thickness never removed the solid infill on slopes, if I'm not mistaken`

It does in every version until this one. 

If after turning off "Ensure vertical shell thickness" in previous versions, you still had unwanted solid fill showing in areas of your model then you knew that they were being recognized by slicer as either top or bottom layers, and slicer was attempting to maintain "Minimum shell thickness".  Lets suppose you had 4 perimeters, In many situations that negates the need for EVST. We as humans need to be smarter then an algorithm, and decide whats best for our prints. 

Alternatively if you needed to remove the rest of this unwanted solid fill you could simply use a modifier to remove those unneeded top or bottom layers.

 `I too wish we could turn solid infill off completely`

You can in any version but this one. Hence my report of not being able to remove\turn off the new implementation of "Ensure vertical shell thickness" 

@AssiveMass

Here is a link that came up recently on the forum that might help you with your solid fill wish :
https://forum.prusa3d.com/forum/original-prusa-i3-mk3s-mk3-how-do-i-print-this-printing-help/how-to-improve-angled-overhang-prints-near-the-bed/

But you'll need to use, well.... not this version.