prusa3d/PrusaSlicer #9958: No z-hop for certain layers causes crashes

Agent-Cain, 2023-03-07:
### Description of the bug

I had some serious layer shifts which I first correlated to some hardware issues. 
[image]
After the second failure I partially disassembled the xy-kinematic and check the hardware, but I got a third layer shift at the exact same height. After that I almost completely disassembled the hole printer and checked every motor, every pully, bearing, belt, screw and everything but couldn't notice a thing.
So I started the print a fourth time and for luck was next to the printer when it almost shifted again. 
This time I could clearly hear something catching. And it was, again, the same height. 
So I tried to get the error on film (see attached) and noticed, that the toolhead (the nozzle to be exact ) catched the infill, because it made no z-hop. Therefore I checked the settings to see if there was something wrong, but z-hop was still set to 4mm. 

That’s where my investigation of the g-code started.
And I was able to find layers without any z-hop (at the height the prints layer shifted).

I’ll try to explain how I found this: 
As the first layer height is set to 0.9mm (printing with an 1.4mm nozzle) and the overall layer height is set to 0.7mm I took a look at some odd random layer: 
[image]
`;LAYER_CHANGE
;Z:350.2`
Was my choice. Therefor the next G1 Z move should be on hop, and it was (G1 Z354.2). In fact there were six hops as there were six G1 Z354.2 in that layer. And there is no layer at height of Z354.2 (because of the 0.7mm layer height).
[image]
So far so fine. So I tried to find some layer without any hop, and I did. 
In fact from height 436.3 to 443.3 there were no hops, so 11 layers in total (436.3, 437, 437.7, 438.4, 439.1, 439.8, 440.5, 441.2, 441.9, 442.6, 443.3). 
After that follows layer at height Z444 with one hop. 

I found some more layers without a hop which I didn’t write down. 
So I think there might be some slicer issued which caused me more than 10kg of PLA waste :')

The model was downloaded from printables:
https://www.printables.com/model/221177-darth-vader-bust-star-wars

And to show, that the printer is running well and can print higher than that vader bust, here is an older print, sliced with PS2.5
[image]

### Project file & How to reproduce

[project.zip](https://github.com/prusa3d/PrusaSlicer/files/10908574/project.zip)

https://user-images.githubusercontent.com/25855585/223404038-40dc3aff-62fa-4c6e-8571-f8d19987747f.mp4

https://user-images.githubusercontent.com/25855585/223405151-7dc41db7-9d23-45e1-8836-8ec4cd217334.mp4

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6-alpha4

### Operating system

Windows 10

### Printer model

Selfbuild 1mx1mx2m voronlike with afterburner

rtyr, 2023-03-07:
You have "Only retract when crossing perimeters" enabled. Therefore all non-retracted travels are also without z-hops. Disable this parameter if you want z-hop/retraction everywhere.

[image]

Agent-Cain, 2023-03-07:
Oh man you made ma day. 
It really could be the solution, because right now I've a print running since 250h and one since 75h where it looks like the parameter is unchecked. There might have sneaked in a mistake on my side while I made the new profile.
Weird is, that the torso in the middle was printed with that option checked and had no problem, that was sliced for sure with the same settings. 

Thank you so far for the answer, I'll have another try and come back with the result.

Agent-Cain, 2023-03-09:
Amazing, as soon as you do it right, it already works
I guess you where right. 
[image]
Thank you!

closed 2023-03-09 by Agent-Cain