prusa3d/PrusaSlicer #9960: Crash during slicing

PrintModz, 2023-03-07:
### Description of the bug

As soon as I want to add 0% infill and a support structure to my object, the program calculates up to 70% and then stops. Here, "Only for support amplifier" was selected.

### Project file & How to reproduce

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0 Alpha 5

### Operating system

Windows 11

### Printer model

Caribou 320

lukasmatena, 2023-03-07:
Please, provide the 3MF project. Thanks.

lukasmatena, 2024-02-18:
No response, closing.

closed 2024-02-18 by lukasmatena