prusa3d/PrusaSlicer #9963: Bridges doesn't respect desired angle print setting

rednaxal, 2023-03-07:
### Description of the bug

Since Alpha 5 the bridge angle setting seems to be working inadequate. Is it because now it detects all kind of bridges as "bridge infill"? 
We need an old way to control the direction of the bridges by providing a degree.

First layer:
[image]

Second layer, bridge angle 1 degree:
[image]

Second layer, bridge angle 2 degrees:
[image]

Two more relevant questions:
Why it doesn't fill some areas (notice the "P" and "R" letters on a second layer)? 
Why is it detected as "Bridge infill" and not just a bridge? 

UPDATE: I found out that it seems like the angle now is not in a degrees but in range [0; Pi], i.e. [0, 3.14]. It would be great to have it in degrees rather than radians.

### Project file & How to reproduce

[2020CorpStrongBug.zip](https://github.com/prusa3d/PrusaSlicer/files/10910315/2020CorpStrongBug.zip)

Slice any model with a bridge and try to control the bridge direction with angle. 

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-alpha5-win64

### Operating system

Windows 10

### Printer model

Kingroon KP3S

kubispe1, 2023-03-07:
> UPDATE: I found out that it seems like the angle now is not in a degrees but in range [0; Pi], i.e. [0, 3.14]. It would be great to have it in degrees rather than radians.

I see it is only in  2.6.0-alpha5. Previous versions looks ok for angles. Thanks

scottmudge, 2023-03-18:
Can confirm, it's in radians.

kubispe1, 2023-03-29:
This issue is fixed and will be part of the next release 2.6.0-alpha6

closed 2023-03-29 by kubispe1