prusa3d/PrusaSlicer #9969: PrusaSlicer-2.6.0-alpha5 - Possible issue with Extended sparse infill

smirgol, 2023-03-07:
### Description of the bug

Comparing alpha4 to alpha5, I've noticed some rather strange behaviour in two spots (screw-holes) when bridging over internal infill:

The bridging does have holes in alpha5 and it looks like it is not always connected to other lines. Where the holes are, there is now infill, which looks to partially be printed mid-air.

In alpha4 there is one solid bridging layer, without any holes.

I assume this is due to the new extended sparse infill feature. I have not tested printing the resulting gcode.

### Project file & How to reproduce

Project File:
[Deckel.zip](https://github.com/prusa3d/PrusaSlicer/files/10913139/Deckel.zip)

Alpha-5:
[image]

Alpha-4:
[image]

Please see Layer 56 at 12.2mm, bottom right screw hole

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.0-alpha5+linux-x64-GTK3

### Operating system

Ubuntu 22.04.2 LTS

### Printer model

MK3S+

MrBlackmidi, 2023-03-12:
lately, i have been noticing a lot of recent issues that are ignored (inc one of mine). hope they are considered tho

KBalage, 2023-04-01:
I can see a similar behaviour - bridge is not calculated and printed correctly - under layer - even if it's not an infill pattern could affect the bridging pattern.

2.6.xx Alpha:
[image]
[image]

2.5.x does a good job with this:
[image]
[image]

I don't mind the new Extended sparse infill method but it should be an optional feature - it would be nice to turn off it's needed.

closed 2026-09-29 by prusaslicer-bot[bot]