prusa3d/PrusaSlicer #9971: Organic Supports generate inside other objects.

jacobcbab, 2023-03-07:
### Description of the bug

If multiple objects i.e. STL files are added onto the build plate, the organic supports do not change to accommodate other parts. Supports are generated for object 1 without any respect to object 2. 

### Project file & How to reproduce

[02. Follower v1.0_2h4m_0.20mm_210C_PLA_ENDER3BLTOUCH.zip](https://github.com/prusa3d/PrusaSlicer/files/10914064/02.Follower.v1.0_2h4m_0.20mm_210C_PLA_ENDER3BLTOUCH.zip)

can be recreated with any two objects that require support.objects

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0 alpha 5

### Operating system

Windows 10

### Printer model

Ender 3

neophyl, 2023-03-07:
Please read https://github.com/prusa3d/PrusaSlicer/issues/9943 (or one of the other many many duplicates of this issue)

kubispe1, 2023-03-08:
Closing as duplicate: #316

closed 2023-03-08 by kubispe1