prusa3d/PrusaSlicer #9973: {Request] Make Counterbores Printable

Merlin246, 2023-03-07:
**Is your feature request related to a problem? Please describe.**
When attempting to print features with inverted counterbores (ie counterbore is facing the build plate), you must use support material or edit the model so it is either:
A - a continuous one layer surface that fills in the hole but can be drilled/punched through.
B - has rectangular cutouts to have two perpendicular bridges before the round hole starts (similar to how Prusa's printed parts are made).

**Describe the solution you'd like**
Add a feature (perhaps with specific painting to enforce/block the setting) to make counterbores printable. This solution should implement option B from above, perhaps with an additional setting to swap to option A.

**Describe how it would work**
Take the counterbored surface, and edit the mesh to include the rectangular cutouts. Either for only one layer or for a specified number of layers per the user.

**Describe alternatives you've considered**
Changing the model in CAD. This can be a PITA especially if prototyping with 3D printing, but having it actually manufactured via CNC or otherwise.

**Additional context**
Example of option B
[image]

Area5142, 2023-03-08:
I often experience the same problem with screw holes that are printed with the countersink side against the building board.

Your solution B is the best in the specific case, but is difficult to model in CAD and a knowledge of layer thickness must be built into the model, which is not desirable. Therefore, I often choose to model a thin layer over the hole(s) (your suggestion A) which after printing can be removed with a drill or a knife.
A single layer that covers the hole(s) and ensures that a bridge can be built is a simple solution that could be used in many other cases - I would like to endorse this solution.

It should be possible in PrusaSlicer to detect all bridge spans that have holes and automatically close them in a single layer - this should of course be an option that can be activated together with the other bridge setups.

mix579, 2023-06-30:
Interesting idea. Prusa itself has implemented option  B in some of its printer parts. I use this method in CAD but it would be great if PS could automatically detect such counterbores and apply this method.

vovodroid, 2024-06-20:
Orca has option A implemented:

[image]

[image]

I also has kind of B option, but I didn't find it useful.
I've tried to port it to Prusa, but without success so far. I got it compiled, but not working.

neophyl, 2024-06-20:
Orca has option A as that was something that was added as an option in Super Slicer.  Although its not automatic, you have to select the correct option.

vovodroid, 2024-06-20:
By a way, I didn't find this option in latest SS 2.5. Does it have another name?

neophyl, 2024-06-20:
@vovodroid 
Print Settings>Perimeters & Shell>Advanced>No perimeters on Bridge Areas.   Set to Fill voids with bridges, or you can also try Keep Only bridges or Keep Bridges and Overhangs.
It is far from perfect and can cause other issues (hence the tooltip warnings) but its where the function got its start.  I dont use Orca so can't say what improvements they may have made to it.

vovodroid, 2024-06-20:
`Set to Fill voids with bridges` works good, Orca does the same.

[image]

I'll take a look on SS code, probably it fits better to Prusa code base.

Sembazuru, 2025-03-11:
### Brainstorming idea for the folks already elbow-deep in the code:
If the automatic filling of unsupported holes (either counterbores or holes on bottom facing faces above the build surface) is becoming problematic, you might be able to borrow the circle selection code from the measurement tool for manually selecting features that need to be filled in with either sacrificial bridging or sequential bridging (borrowing the terms from the Maker's Muse recent [video](https://www.youtube.com/watch?v=KBuWcT8XkhA) about adding these features in CAD). Use the circle interpolation selecting routine from the measurement tool and then apply either sacrificial bridging over the hole or sequential bridging geometry modification to the surface that is co-planer to the selected hole (again there may be routines in the measurement tool to borrow to figure out the co-planer surface to the selected hole).

### Justification for management:
I think PrusaSlicer should have this feature because one might not have access to the original CAD file (or an easily modifiable STEP file) when downloading a model that would benefit from these features. I currently manually close holes by adding one layer tall cylinder parts in the slicer, but that is tedious to do. Especially getting the bottom of the disk co-planer with the surface around the hole. Maybe adding object snap features would be useful for this and other user operations, but that would be a completely separate GitHub issue.

girak49, 2026-09-02:
Can anyone confirm if this feature is in the newest build of PS 3.x? I still support this feature request.
It's not available on linux yet AFAIK

Sembazuru, 2026-09-02:
I don't see this feature yet. Not surprised as Prusa's main focus right now is (and should be) making sure all the features of 2.x is ported over to 3.x. That is an ongoing effort.

But, there is a plugin API using Lua as the scripting language. I haven't found documentation for the API yet so I don't know if the hooks for identifying features on a loaded STL are in the API. But the API is currently a moving target (will probably change minorly to significantly between alpha releases). Remember "Alpha Release" = "Here be Dragons"

girak49, 2026-09-02:
> I don't see this feature yet. Not surprised as Prusa's main focus right now is (and should be) making sure all the features of 2.x is ported over to 3.x. That is an ongoing effort.
>
> But, there is a [...]

Thanks, I didn't want to assume it wasn't added.