prusa3d/PrusaSlicer #9976: + buttons cut off in some spots under print settings

Jakeo-93, 2023-03-08:
### Description of the bug

Strange bug I'm getting on 2.5 and even the new 2.6.  Appimages,  As stated in the title in some spots under the print settings tab my plus icons are cut off.  Not sure whats causing it as my laptop runs the same Os and the issue is only on my desktop . Could be my graphics driver but I'm not sure 

### Project file & How to reproduce

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer2.5 Appimage 

### Operating system

Endeavour Os

### Printer model

Ender 3

lukasmatena, 2023-03-08:
Already reported in #8028.

lukasmatena, 2024-02-19:
Closing as a duplicate of #8028.

closed 2024-02-19 by lukasmatena