prusa3d/PrusaSlicer #9977: Toggle for Thick bridges is not working

Snuff1eupagus, 2023-03-08:
### Description of the bug

Thick bridges is always on, ticked or un-ticked. 

### Project file & How to reproduce

Create a cylinder slice with default settings, observe bridging. Note : height is printed at 0.390.
Tick Thick bridges, re-slice. Observe bridging. Note : height is printed at 0.390 no change.

Width does not change either however height shows the problem with better visual contrast. 

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6 alpha 5

### Operating system

Win 7 - 10

### Printer model

n/a

kubispe1, 2023-03-09:
Please see https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha5 section 'Extend sparse infill'
> The bridge infill is now always using 'Thick bridges'. 

If I understand your statements correctly this is the reason.
I am closing issue. If the point of your issue is different, please attache 3mf and let's discuss it more deep.

closed 2023-03-09 by kubispe1

Snuff1eupagus, 2023-03-09:
So I'm to understand that users no longer have Control over the Angle and Type of Internal bridging because of the development of 'Extend sparse infill' ?