prusa3d/PrusaSlicer #9978: Alpha5: Mission perimeters on upper sloped surface

sethml, 2023-03-08:
### Description of the bug

An object with a shallow slope to the top surface and only 2 perimeters has gaps in the top layer, exposing the infill inside - see attached .3mf and screenshots. I've tried to replicate this with a simpler object and failed, so perhaps it has something subtle to do with the geometry of the object. The problem does not occur with 2.6.0-alpha4 - the gaps contain solid infill. Thus I suspect this is related to the "Ensure vertical shell thickness improvement". However, I would have expected that having "extra perimeters if needed" checked would have filled in these gaps with extra perimeters.

I do note that I have "minimum layer thickness" set to 0, but there's a little tip below reading "Top shell is 0.4 mm thick for layer height 0.2 mm. Minimum top shell thickness is 0.16 mm.\nBottom shell is 0.4mm thick for layer height 0.2mm. Minimum bottom shell thickness is 0.16 mm.". I'm not sure how to read this, but it seems like it's saying that a value of 0 is interpreted as 0.16 mm. I think it could be worded more clearly. In any case, setting top minimum layer thickness to 0.16 mm makes no change, but setting it to 0.2 mm (one layer height) causes solid infill to be added to the gaps.

I'm not sure what the best solution is to this problem, but leaving gaps in the top surface with the default settings seems poor. I think a good solution would be to have "extra perimeters if needed" add extra perimeters on the top surface to fill the gaps. Another solution would be to have the minimum shell thickness be at least one layer height (0.2 mm in this case).

### Project file & How to reproduce

[Borken Top Surfaces.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/10917713/Borken.Top.Surfaces.3mf.zip)

With 2.6.0-alpha5:
[image]

With 2.6.0-alpha4:
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.6.0-alpha5+MacOS-arm64-202303061441

### Operating system

System Version: macOS Ventura Version 13.2.1 (Build 22D68)

### Printer model

Elegoo Neptune 3 with Kustom Klipper

kubispe1, 2023-03-08:
Hi, I can confirm this issue appears in 2.6.0-alpha5 and I would also guess it is related to ensure "Ensure vertical shell thickness improvement"
This issue not appear in 2.6.0-alpha4
We will look into it. Thanks

When you apply in your case: 
[image]

There are no holes in upper  top solid infill
[image]

kubispe1, 2023-03-10:
SPE-1587

FidelCapo, 2023-04-14:
This issue should be fixed and fix will be part of next release.
Closing.

closed 2023-04-14 by FidelCapo