prusa3d/PrusaSlicer #9986: Cut Tool - Set male and female sizes independently

myoung34, 2023-03-08:
**Is your feature request related to a problem? Please describe.**

When using the cut tool (which is amazing) you can only set the single size for the male and female connectors

Im printing something large and using a 1.2mm nozzle, so tolerances aren't great

**Describe the solution you'd like**

It would be ideal if i could set the sizes independently via an opt-in checkbox (like how resizing has a "lock" feature so they stay 1:1)

IE: set the male connector to 9mm and the female connector to 10mm

**Describe how it would work**

In the connector sizing portion of the UX:

* 2 size boxes as opposed to 1 labeled `Male` and `Female` (or recepticle/socket, etc)
* a "lock" icon that is on by default so current behavior is preserved. Setting one keeps the other in sync (no change in behavior)
* Hitting the lock button to unlock it lets you toggle them independently

**Describe alternatives you've considered**

I tried to fix my current issue with heat and it did not work well

lukasmatena, 2023-03-08:
[image]

myoung34, 2023-03-08:
Weird, I definitely did not see this. Ill make sure I have the appropriate version.

myoung34, 2023-03-08:
OK so i do have this. Is it possible to add some tooltips? I had no idea tolerance was what I was looking for :smile:

Jan-Soustruznik, 2024-03-11:
@myoung34 , thank you for your feedback, but I don't know if it is necessary to further explain what "tolerance" is. It is a standard term.

closed 2024-03-11 by Jan-Soustruznik