prusa3d/PrusaSlicer #9988: [2.6.0-alpha5] Extend top layer of bottom infill at least into the perimeters

Itox001, 2023-03-08:
### Description of the bug

Thanks for the amazing improvements in the latest alpha, this things were long awaited and help a lot with a lot of situations. Having said that, I noticed a strange behavior with the new way to generate infill. Now, the top layer of bottom infill is only covering the infill area of the corresponding bottom surface and no more, which results in weaker and outright failing prints in the right circumstances. This doesn't sound too clear so I'll post screenshots:

2nd layer of Benchy, sliced with 1 bottom layer on the alpha4:
[image]

2nd layer of Benchy, sliced with 1 bottom layer on the alpha5:
[image]

You can see that now, the top last (and only, in this case) layer of bottom infill got generated exactly on top of the embossed text, which results in bridges that are not anchored to the perimeters. While using much less material, this is just going to fail. Therefore, it should extend at least into the perimeters of the surface for the bridges to be anchored, I don't know if even a bit further would be beneficial like if happens when you slice it with 2 bottom layers, where the first one covers the text plus some more:
[image]

This happens always with the top layer of the bottom infill. This will not fail since it can be printed already on top of the first bottom layer, but will probably not help at all with print strength:
[image]

The above is I'm pretty sure a bug, but I have also noticed another thing that I want to bring to your attention which is less clear. The first layer of solid infill over sparse infill seems to be a bit tight, and some little corners of the second layer of solid infill seem a bit unsupported. This is pretty small though, but I'm not sure if this will cause issues in some edge case and could simply be addressed by expanding the first solid layer over sparse infill a bit:
[image]
[image]

Thanks for the good work!

### Project file & How to reproduce

[BottomInfillBug3DBenchy.zip](https://github.com/prusa3d/PrusaSlicer/files/10923962/BottomInfillBug3DBenchy.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0-aplha5

### Operating system

Windows 10

### Printer model

Sovol SV01

DFliyerz, 2023-03-21:
I'm running into this issue too, I'm printing with only perimeters for a first layer test and the second layer bridge ends up like this:

[image]

Itox001, 2023-04-02:
On the alpha6 its even worse, now even the X is not completely filled:
[image]

kubispe1, 2023-05-17:
The improvement in the Ensuring algorithm should resolve the issue with non-anchored bridges. Please refer to the changelog of **PS 2.6.0-beta1** for more details. If you encounter any persistent issues, please let us know. Thank you.

closed 2023-05-17 by kubispe1