prusa3d/PrusaSlicer #9989: Show model and support material separately

guido66611x, 2023-03-08:
**Is your feature request related to a problem? Please describe.**
I often need to know how much an actual printed model will weigh, not including the support material, since the supports will be removed and discarded (and are not part of the final product).

**Describe the solution you'd like**
The bottom right "Sliced Info" window currently lists "Used Filament (g)", and instead it could say "Used Filament (g) Model / Supports / Total:  200 / 125 / 325", or perhaps have a tooltip that pops up when you click on the simpler "Used Filament (g)" entry.  It could also be a popup on mouseover, however, I often need to write these numbers down for total-assembly-weight reasons, so I need the window to stay visible while I move the mouse elsewhere (e.g. Excel).

**Describe how it would work**
From the legend window, it looks like the slicer is already calculating each different type of filament with enough granularity that it has all the necessary info to easily sum up the model and support weights separately.

**Describe alternatives you've considered**
Currently, I slice it once with supports turned off, and write down the mass.  Then I reslice it with supports turned on, to actually export the g-code.  I could also add up the line entries in the legend, but it's less work, and less error prone, to just reslice.  However, on large models, slicing sometimes takes a long time (e.g. 2 minutes).

**Additional context**
N/A

lukasmatena, 2023-03-08:
This makes some sense. We are already showing the amount for the wipe tower separately.

rcmaniac25, 2026-09-13:
Would be nice to have.