prusa3d/PrusaSlicer #10000: Export as STL/OBJ with issues

JWartenberg, 2023-03-09:
### Description of the bug

I would like to export a file as STL-File.
I select export as STL/OBJ via right-click. If I click the save-button without further interaction (<filename>.STL) nothing is saved, no error message will apear. It happens simply nothing.

Same procedure as above, but if changing the suffix of the filename form STL to stl (upper case to lower case) it works.

### Project file & How to reproduce

s, above

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0 Alpha 5

### Operating system

windows 11

### Printer model

MK3S+ with MMU2

kubispe1, 2023-03-10:
I can reproduce it in 2.6.0 Alpha 5
- It works for 2.5.0

kocikdav, 2023-03-10:
Should be soled in next alpha.
Commit c2fe61261dd4490b63114c8154575e036617c9ec

closed 2023-03-10 by kocikdav