prusa3d/PrusaSlicer #10001: [2.6.0-alpha5] New solid infill too dense resulting in deformations

NielschB, 2023-03-09:
### Description of the bug

The new solid infill pattern seems to be quite dense and easily resulting in over-extrusion in some areas which in turn causes deformation in following layers.  The indicated areas in the screenshow below seem to be prone to over-extrusion. Additionally the generated paths in the corners look like they could be optimized.

[image]

### Project file & How to reproduce

[afmetingen_test_30x10x20.zip](https://github.com/prusa3d/PrusaSlicer/files/10933972/afmetingen_test_30x10x20.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-alpha5-win64

### Operating system

Windows10

### Printer model

Prusa MK3S

NielschB, 2023-03-09:
Photo of the actual print after the second layer
[image]

NielschB, 2023-03-09:
Similar problem mentioned in #9962

jwkicklighter, 2023-03-10:
I am experiencing a similar issue. Though it is not obvious in the slicer preview, excess material is causing deformations at the top of my prints.

[image]

BillySp33d, 2023-04-04:
This is still a problem in alpha6. The new infills always over extrude for me

NielschB, 2023-04-05:
Same here, due to this issue the Prusacliser 2.6.0 alpha 5 and 6 releases are not usable for me.

jwkicklighter, 2023-04-14:
Curious, what nozzle sizes are you all using? I'm noticing this problem more prominently with 0.6mm nozzle and have had successful prints with 0.4mm nozzles.

NielschB, 2023-04-14:
0.4mm nozzle

NielschB, 2023-07-05:
This issue seems resolved with 2.6.0 release as the fill patterns have been reverted back to original

closed 2023-07-05 by NielschB