prusa3d/PrusaSlicer #9992: Export Plate Including Supports Grayed Out

dmbaio, 2023-03-09:
### Description of the bug

In alpha 5, on macOS, the option to export an FDM plate including supports is not available.

[image]

### Project file & How to reproduce

[Azalea.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/10927312/Azalea.3mf.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-alpha5+MacOS-arm64

### Operating system

macOS Ventura 13.3

### Printer model

AnkerMake M5

kubispe1, 2023-03-09:
Export with plate and supports is enabled only for SLA. That's the reason, why it is grayed Out. Could be as an FR, but at the moment we don't plan to enable it. 
What could be the purpose of this export?

dmbaio, 2023-03-09:
My use case for wanting to be able to do that for FDM prints so far has been when Prusa-sliced files continuously fail but I still want to use the Prusa-generated supports, and would need to have them included in the same STL with the model to use in another slicer. That particular model, actually, when sliced with this alpha version of Prusa, failed every time I tried to print it no matter what I tweaked. Opening the same .3mf file in Cura and slicing it there printed perfectly, although I did have to add a support-enforcing cube due to not being able to paint on supports at the base.

TheW00Xer, 2023-03-13:
I have similar issue with Prusa-sliced files failing but I also want to use Prusa generated supports so adding this function for FDM printers would be a huge help.

secv, 2023-09-24:
I would use this feature with the intention of further editing the exported file in Blender.

closed 2026-09-29 by prusaslicer-bot[bot]