prusa3d/PrusaSlicer #9994: A Tool Change is triggered even though I assigned everything to extruder 1

Bumblebiber, 2023-03-09:
### Description of the bug

There's a tool change where should be none. PS apparently wants to print the text with extruder 2. So it changes and tries to print but extr.2 is not heated and there's another filament loaded.

### Project file & How to reproduce

[image]

[Retraction_test.zip](https://github.com/prusa3d/PrusaSlicer/files/10930271/Retraction_test.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-alpha5+win64

### Operating system

Windows 11

### Printer model

EvoTech EL-102

lukasmatena, 2023-03-09:
@Bumblebiber This part of the gcode is from your custom start gcode. PS doesn't "want" anything, it just pasted the value from the config.

Bumblebiber, 2023-03-09:
> @Bumblebiber This part of the gcode is from your custom start gcode. PS doesn't "want" anything, it just pasted the value from the config.

Sure, but

Bumblebiber, 2023-03-09:
> @Bumblebiber This part of the gcode is from your custom start gcode. PS doesn't "want" anything, it just pasted the value from the config.

It's custom GCode, yes. But not Start GCode. It's the tool change GCode that should only be triggered for a tool change, which should not happen in this file.

lukasmatena, 2023-03-09:
@Bumblebiber Ok, I'll check again later. Can you please post the 3mf (not just the gcode)? Thanks.

themelle, 2023-03-09:
> It's custom GCode, yes. But not Start GCode. It's the tool change GCode that should only be triggered for a tool change, which should not happen in this file.
> 
> This happens multiple times throughout the print, so it's not a start gcode issue...

You selected T0 (Extruder 1) for your print objects. So PS makes a tool change to ensure T0 is used.

I think setting extruder to "Standard" (not "1" tor "2") should output Gcode without an initial tool change, as its using whatever tool is already active.

Bumblebiber, 2023-03-10:
> You selected T0 (Extruder 1) for your print objects. So PS makes a tool change to ensure T0 is used.
>
> I think setting extruder to "Standard" (not "1" tor "2") should output Gcode without an initial [...]

That's how it was set by default. So I tried to assign Extruder 1 manually, which resulted in no change.

Bumblebiber, 2023-03-10:
[stringing_test.zip](https://github.com/prusa3d/PrusaSlicer/files/10940669/stringing_test.zip)

Bumblebiber, 2023-03-10:
[stringing_test_wothout_text.zip](https://github.com/prusa3d/PrusaSlicer/files/10940777/stringing_test_wothout_text.zip)

Same project but without text - and the tool change is gone

lukasmatena, 2023-03-10:
@Bumblebiber The custom G-Codes don't make the debugging easy, but there are several things I noticed: the text part is floating above the object, the supports are enabled, support_material_interface_extruder is set to 2. I believe it has some significance.

Bumblebiber, 2023-03-10:
> @Bumblebiber The custom G-Codes don't make the debugging easy, but there are several things I noticed: the text part is floating above the object, the supports are enabled, support_material_interface_extruder is set to 2. I believe it has some significance.

Damn, you're right. But it's a new file I just made to reconstruct yesterday's case. Maybe there it was the same reason. Though it should not trigger a tool change, should it?

lukasmatena, 2023-03-10:
@Bumblebiber The gap triggers supports. If there is a specific extruder set for printing supports, that may trigger a toolchange.

closed 2026-09-29 by prusaslicer-bot[bot]