prusa3d/PrusaSlicer #9996: Crashing

ljungis, 2023-03-09:
### Description of the bug

When I slice the attached file as it comes into the slicer it works ok, but if I turn it 90 dgr by using the lay flat comand the program crashes when I press the slice button.

### Project file & How to reproduce

[Clamp-25.zip](https://github.com/prusa3d/PrusaSlicer/files/10932414/Clamp-25.zip)
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-alpha5+linux-x64-GTK3-20

### Operating system

Linux Mint 20.3 Cinnamon

### Printer model

Wanhao Duplicator i3 Plus MK2

kubispe1, 2023-03-09:
On Win 10 there I get Access violation.
Thanks

ljungis, 2023-03-09:
Thanks for your reply. I attach the zip file again, but as password password  protected.
Try to open it using the password "clamp"
[Clamp-25.zip](https://github.com/prusa3d/PrusaSlicer/files/10933494/Clamp-25.zip)

bubnikv, 2023-03-09:
access to a null ptr layer
[image: image.png]

ljungis, 2023-03-09:
Thank's for your reply bubnikv, but I don't understand what "access to a null ptr layer" means.

lukasmatena, 2023-03-10:
This is already taken care of (SPE-1590). It will be fixed in the next release.

closed 2023-03-10 by ljungis