prusa3d/PrusaSlicer #9999: [2.6.0-alpha5] Bridges do not overlap perimeters which result in failed print

NielschB, 2023-03-09:
### Description of the bug

Bridge generation seems to stop the bridge at the external perimeter instead of overlapping which results in a failed print. This behaviour was not present in PrusaSlicer 2.6.0-alpha4.

Screenshots to illustrate the issue below

[image]

[image]

I would expect the bridges to at least overlap the external perimeters, but instead it stops just before them.
Actual prints show failing bridges at this location.

### Project file & How to reproduce

[afmetingen_test_30x10x20.zip](https://github.com/prusa3d/PrusaSlicer/files/10933785/afmetingen_test_30x10x20.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-alpha5-win64

### Operating system

windows10

### Printer model

Prusa MK3S

NielschB, 2023-03-09:
Similar bridge problem is reported in #9988

CloudYeti, 2023-05-12:
I am experiencing the same issue with 2.6.0 alpha 6. There is not enough overlap for the bridging to anchor to the perimeter and it results loops that are not attached.

RickM777, 2023-06-22:
Same issue still in 2.6.0 signed.  At least in my case where one side is only 2 perimeters wide.  Screenshot showing 2 layers, the infill layer and the one immediately below.  You can see at the top it goes over the perimeter but on the bottom it only goes next to it.  Picture shows result.  This was not an issue for these models in any prusaslicer version before 2.6.
[image]

[image]

ddacunha, 2023-09-12:
I have very similar issue, it's obvious when comparing 2.6.1 to 2.5.2

[image]
[image]

zanderpu, 2023-09-14:
I found that this problem can be solved by setting the layer height. For example, if the printing layer height is 0.2mm, if the top height of the bridge is only 0.2mm, the bridge will fail. If the top height of the model is higher than 0.2, the bridge will be successful, or If the height of the top layer of the model cannot be changed, the bridge can be successfully built by setting the layer height to 0.1mm. The version of PrusaSlicer I use is 2.6.1

[image]
[image]
[image]

alexeyzel, 2024-02-18:
I think I have the same problem, here bridge does not overlap with bridging perimeter and falls down.
[image]

Jan-Soustruznik, 2024-02-23:
Hello, @alexeyzel , Could you share a 3MF file, please? Thank you

alexeyzel, 2024-02-25:
I have only STL file, here it is (uploading in ZIP archive as STL extension is not supported)
[calicat.zip](https://github.com/prusa3d/PrusaSlicer/files/14396757/calicat.zip)

brandonbeaumia, 2024-03-12:
Here's an object with with the same issue. Using 0.16mm as the layer height bridges appropriately. Using 0.2mm doesn't.

Project file: 
[Shot Counter Watch-Body.zip](https://github.com/prusa3d/PrusaSlicer/files/14579677/Shot.Counter.Watch-Body.zip) (rename to .3mf)

0.20mm (top solid infill hidden):
[image]

0.20mm (top solid infill shown):
[image]

0.16mm (all feature types shown):
[image]

It's very frustrating to not be able to bridge just a few millimeters. :disappointed:

lukasmatena, 2024-03-21:
Fixed in [PrusaSlicer 2.7.3-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.3-alpha1). Closing.

closed 2024-03-21 by lukasmatena