prusa3d/PrusaSlicer #10004: part cooling fan ramp-up time

jakub874, 2023-03-10:
**Is your feature request related to a problem? Please describe.**
On faster printers the changes in part cooling fan % lag behind when they are actually needed since fans have rotational inertia and can't spin up instantly. Say you're printing at 30% fan then have a bridge feature that requires 100% fan. Allowing the fan to ramp-up to the speed before it hits the feature would let the actual feature be at said the requested speed.

**Describe the solution you'd like**
An option for a time ramp-up in seconds before a feature with a fan increase needs it (ex. bridge).

**Describe how it would work**
Fan is fed with 100% PWM for the time specified in the field in the slicer. Scale so that smaller increases have a shorter ramp up time, with the value in the slicer specifying the 0-100% ramp-up time.

**Describe alternatives you've considered**
no alternatives possible

**Additional context**
no additional context

CanineTunic, 2023-03-10:
good idea

fry99999, 2023-03-10:
real

neophyl, 2023-03-10:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/3590

jakub874, 2023-03-13:
> Could possibly be a printer setting of of acceleration/deceleration for the fan. That way it can be calculated when to start the fan up to reach the right point, and deceleration as an optional "do we actually emit a fan speed adjustment gcode? Or will we get to another fan use point?" calculation. At the furthest end, maybe it even decides some pathing as there are parts that will curl if the fan is on and the fan won't be off in time. Doesn't even need to be fast prints, just a bunch of small prints.

yea that's basically what it is, advance the fan PWM enough so that it's at the right speed when it hits the bridge. in terms of deceleration it shouldn't be an issue for anything afterwards, a split second of fan wont do anything.

jakub874, 2023-03-13:
> Duplicate of #3590

vastly different implementation, i wouldn't really call it a duplicate. pausing the print head is silly, it allows for unnecessary ooze and if anything makes the "overcooling" concern mentioned worse, if it was a concern at all already.

pmbdk, 2023-04-01:
I second this. Although bridges are certainly a problem, overhangs are actually worse imo. I usually print with no cooling on larger prints as this improves layer adhesion _a lot_ but for overhangs (using the new dynamic speed settings in 2.6) there is no ramp up, so there is virtualy no impact of those settings on overhangs which are basically already printed once the fan has started.

janroos, 2023-04-30:
I guess I have a fan with very slow spinup speed, so I have dreamed of such a feature for a long time. The first crosses of the bridge sack hugely, making the print quality worse than it could be.

unameisfine, 2023-05-21:
Implement it plz. But in a different form. With inertia paramater and calibration sample.

clintonefisher, 2023-10-22:
Yes please!  This is really important with TPU bridges, which are difficult to begin with.
With TPU (NinjaFlex in my case) we have to significantly reduce the volumetric flow rate for bridging to stretch the filament to prevent sag.  The problem with the fan-lag is that the first part of the bridge stays soft, and then gets pulled away from the side by the rest of the bridge, resulting in a broken bridge strand.   If the fan is up-to-speed before the bridge starts, the bridge-strand is more  likely to survive.

I have set the fan higher with M106 for a full layer where I know it is mostly bridges, and the difference is huge.  Bridge strands complete instead of break.

Fan-spin-up compensation would be fantastic!

I would have a checkbox to enable the feature, with offset and multiplier values to account for different fans, with the standard Prusa MK3 fan being the default.

amicloud, 2023-12-25:
This is a desperately needed feature.

jakub874, 2023-12-27:
> I would have a checkbox to enable the feature, with offset and multiplier values to account for different fans, with the standard Prusa MK3 fan being the default.

As implemented in superslicer should be adequate, where it's simply a "fan ramp up time" field where you enter the time it takes for the fan to spin up from 0-100. 

Then as the tool approaches a feature that calls for a new PWM setting it could set the fan PWM to the new PWM setting in advance of the feature by the same ramp up time as input into the field to allow the fan to approach the correct speed.

I think superslicer sets the PWM to 100% for a fraction of the time of its "ramp up time" field based on the difference in fan speed it's approaching, and then probably has an internal timer/limit for when it can apply ramp up time in the case that there are alternating features where the fan would potentially otherwise spin too fast.

I think the second option (as done by superslicer but with potentially a few tweaks) could be better as it uses the known value of fan ramp up while PWM is at 100%, whereas at lower PWMs the fan make take longer to ramp up. I'm leaving both up as food for thought.

FMachado84, 2024-03-17:
I agree. This needs to be added into PrusaSlicer with roughly the same implementation that SuperSlicer contains. I run a 4028 fan and it does not spool up as quickly as a 5015. This means that most smaller bridges, as well as large sections of overhangs don't receive sufficient cooling while the fan is spooling up to full speed.

Hopefully the devs see this and consider the importance of this feature for many users.

Thanks!

Jan-Soustruznik, 2024-07-10:
Closed as duplicate of https://github.com/prusa3d/PrusaSlicer/issues/10130

closed 2024-07-10 by Jan-Soustruznik