prusa3d/PrusaSlicer #10005: Prusaslicer MSLA profile not exporting properly

CAPNCRACKSPARROW, 2023-03-10:
### Description of the bug

I love prusaslicer and its my only slicer i use for all my fdm machines, i was really excited when i saw alpha 2.6 had a profile in the config wizard for the photon mono x. When using it, it will export the masked sla file, i select the .pwmx option but the mono x does not read the file and says its invalid. 

### Project file & How to reproduce

[Photon mono x pwmx problem.zip](https://github.com/prusa3d/PrusaSlicer/files/10939007/Photon.mono.x.pwmx.problem.zip)
exports to usb just fine.
Machine does not recognize or preview file.
Unable to print from this.
other program lychee slicer using same orientation and file works fine.

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0 Alpha 5 win64

### Operating system

windows 11

### Printer model

Photon Mono x

jmd73, 2023-03-10:
@CAPNCRACKSPARROW Maybe [PR 9929](https://github.com/prusa3d/PrusaSlicer/pull/9929)?

tamasmeszaros, 2023-03-10:
Hello! I think the pwmx format was accepted as a PR from @ole00. If he could have a look at the issue as well as the new PR #9929 that would be a great help.

jmd73, 2023-03-10:
@tamasmeszaros It looks like @ole00 has given his OK for PR #9929 in #7917.

Mimoja, 2023-03-10:
I dont have a Mono X and can sadly not test it. My mono prints fine with my PR which is absolutely not helpfull at all :/

ole00, 2023-03-13:
I do have Mono X, I tried to compile latest PrusSlicer to test the latest changes (pr9929 rebased on top of version_2.6.0-alpha5). Unfortunately I can't build it any more - the cmake system can't finish configuration on my Linux system (Lubuntu 18.04, cmake 3.25.3).  Here is the log with the errors:

-- EXPAT found in /usr/local/lib/cmake/expat-2.4.8
CMake Error at CMakeLists.txt:448 (add_library):
  add_library cannot create target "libexpat" because another target with the
  same name already exists.  The existing target is an interface library
  created in source directory "/home/ole/dev/PrusaSlicer_pr9929".  See
  documentation for policy CMP0002 for more details.

-- Falling back to MODULE search for QHULL...
CMake Error at src/CMakeLists.txt:19 (add_library):
  add_library cannot create target "qhull" because another target with the
  same name already exists.  The existing target is an interface library
  created in source directory "/home/ole/dev/PrusaSlicer_pr9929/src".  See
  documentation for policy CMP0002 for more details.

-- Could NOT find NanoSVG (missing: NanoSVG_DIR)
  Note: my nanosvg is located in /usr/local/include/nanosvg/nanosvg.h - so quite standard directory

ole00, 2023-03-14:
I've tested PR #9929 rebased on top of version_2.6.0-alpha5. So far it seems to work fine:
- no PR #9929 specific compilation/link issues were found
- Printer selection wizard works OK, correct printer and printing/material profiles are selected
- slicing and export works without errors
-  exported file has .pwmx extension (correct for Photon Mono x)
-  exported file can be opened and verified in PhotonWorkshop. I also checked the file contents/details - they seem to match the expected values for .pwmx file
-  exported file has a correct thumbnail
-  the stone_new.pwmx (exported by this PS version) is recognised by my Photon Mono X file and is previewed
- 
I have not printed the file so far, it will have to wait until this weekend. I'll report my print results.

@CAPNCRACKSPARROW : please check your printer firmware version is the same as mine (see the picture bellow, the exported file stone_new.pwmx is recognised by the printer)

[image]

ole00, 2023-03-19:
I did a sample print test from the .pwmx file exported by the new version of PS (PR 9929 rebased on top of version_2.6.0-alpha 5) on my Mono X and all seems OK. I printed a couple of small brackets, nothing fancy - but it proves the point. Hope it helps.
[image]

kubispe1, 2023-05-17:
According to https://github.com/prusa3d/PrusaSlicer/pull/9929#issuecomment-1472596826, I am closing this issue. If any persistent issues arise, please let us know.

closed 2023-05-17 by kubispe1