prusa3d/PrusaSlicer #10008: More Per-Object Settings

Zettymaster, 2023-03-10:
**Is your feature request related to a problem? Please describe.**
When comparing print settings side-by-side, like when tuning new types of filament or optimizing a print profile, it is often beneficial to print the same object multiple times with slightly different settings. While some of these settings are already available, I would like to see more (see below for specific mentions) to make test files that may be used and reused multiple times - possibly on different printers. Currently I would need to forego the new features in PrusaSlicer for the configuability that SuperSlicer currently has the edge in. This has the obvious issue that both slicers have different behavior that makes the result possibly not comparable.

**Describe the solution you'd like**
Adding more per-object settings, namely:
- Infill before Perimeters 
- Acceleration
- Auto Speed
- Pressure Equalizer
- Filament Settings (the most important one and the one why I am even making this feature request)
  - Extrusion Multiplier
  - Nozzle Temperature
  - Cooling
    - Fan speed min/max
    - Cooling thresholds

**Describe how it would work**
Since I have no idea on how the code works, I'll just say they work on a per-object basis...
Filament settings could use their own spot in the menu, while Auto Speed and Pressure Equalizer would be most at home in the Speed settings.

As for the temperature settings: SuperSlicer seems to implement them as an Offset, which I find a little unwieldy, absolute values would be nice here.
Same story for EM: SS uses a percentage (presumably of the normal EM set in the Filament Settings), I would find an absolute easier to use, since I won't have to do math in my head as much when its absolute.

**Describe alternatives you've considered**
- Exporting my current config
- Importing said config in SuperSlicer
- Set per object settings
- Slice in SuperSlicer
- Hope everything works the same in both Slicers...

strayr, 2023-05-10:
Anything we need to calibrate with a stack of test objects should be available per object.

The thin-wall vase method doesn't work for ABS, 5 top layers over 5layers of infill test objects work well, it's infuriating printing them one at a time, so I'm using SuperSlicer to calibrate and copying settings across.

GarthSnyder, 2023-06-05:
I strongly support this proposal, particularly for filament values that require tuning: extrusion multiplier, nozzle temperature, cooling params, pressure advance.

In the past I've been using SuperSlicer for this, but it doesn't seem to be under active development these days. I'd really like to be able to do these calibrations in PrusaSlicer.

Tupson444, 2026-09-12:
still relevant