prusa3d/PrusaSlicer #10011: Organic supports: allow increasing number of support build plate disc layers

sethml, 2023-03-10:
**Is your feature request related to a problem? Please describe.**
With organic supports the first layer is a disc with the organic tube printed on subsequent layers. The disc is very helpful for bed adhesion - sometimes too helpful. It can be really difficult to remove the discs from the bed. Adding a setting to make the discs 2 or more layers thick would make them much more removable.

**Describe the solution you'd like**
A new setting, something like "Support build plate disc layers" or something that could be increased from 1 to 2 or 3.

Here's what my build plate looks like at the moment after printing a bunch with organic supports:
[image]

jypma, 2023-03-12:
I believe the default layers for PrusaSlicer's skirt is already also 2, for the same reason.

vovodroid, 2023-06-13:
Another reason is exactly opposite - weak adhesion and wrapping.

closed 2025-08-05 by Jan-Soustruznik