prusa3d/PrusaSlicer #10013: Gcode for Z is missing leading zero, and possibly extraneous usage

tyeth, 2023-03-10:
### Description of the bug

The latest beta has some Z elements of GCode that are missing the leading zero. I don't think my printer likes this and I've not seen it before. You'll notice the comment mentions Z0.3 but the commands use `Z.3`
There also seem to be loads of repeatitions in the first layer of `Z.3` as part of other gcode commands, but no Z height change ever occurs so I consider these extraneous. 

[image]

### Project file & How to reproduce

Using snapmaker 2.0 A350, petg, varying Z offset between -0.2 and 0.0
[120MM Fan to Filter duct - Main Duct (1).zip](https://github.com/prusa3d/PrusaSlicer/files/10946544/120MM.Fan.to.Filter.duct.-.Main.Duct.1.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6b5

### Operating system

win11

### Printer model

Snapmaker A350

rtyr, 2023-03-10:
Intended behavior, introduced long time ago (I believe in 2.4.0)
See https://github.com/prusa3d/PrusaSlicer/issues/6996.

closed 2023-03-10 by rtyr

tyeth, 2023-03-11:
Thanks @rtyr . It's still a problem, and I don't believe it's been addressed, merely cast aside. You'll notice the label Awaiting Response on that thread, and it's full of valid points about breaking backwards compatibility for no real gain.