prusa3d/PrusaSlicer #10018: Using start g_code

mumrikmamrik, 2023-03-11:
### Description of the bug

I'm using prusaslicer with Klipper and start g-code is written in Klipper and in prusaslicer start g-code is only 
"START_PRINT EXTRUDER_TEMP=275 BED_TEMP=105".
Prusaslicer (i suppose) parce my start g-code, found that i'm not heating extruder and bed and decide to do this itself without any exception. 

btw, add klipper config. 

### Project file & How to reproduce

add START_PRINT EXTRUDER_TEMP=275 BED_TEMP=105 to any start_code and delete any other start preheat code

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0-alpha5

### Operating system

macOS 12.6

### Printer model

ender3, klipper

NielschB, 2023-03-12:
Perfect! What else would you expect? Again, great detailled description of your problem/solution, keep wasting everybody's time!

Gunner087, 2023-03-18:
@mumrikmamrik this can be worked around for now by adding "M190 S0" and "M109 S0" commands to the very beginning of your start gcode. This prevents PrusaSlicer from automatically adding the temperature commands after the start gcode.
[image]

lukasmatena, 2023-04-02:
Setting temperatures automatically is optional since [2.6.0-alpha6](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha6). Closing.

closed 2023-04-02 by lukasmatena