prusa3d/PrusaSlicer #10019: Add "complicated" setting of overhang speed into expert mode. It is very useful for extreme overhang printing.

mumrikmamrik, 2023-03-11:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe how it would work**
Be as descriptive as your skill level allows you.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or paint drawing about the feature request here.

NielschB, 2023-03-12:
Quite detailed description of the problem/solution, keep up the good work!

Gunner087, 2023-03-18:
I also opened an issue (#9950) about this as soon as the ability to define the overlap % was removed with alpha 5.

closed 2026-09-29 by prusaslicer-bot[bot]