prusa3d/PrusaSlicer #10020: Synchronising settings

diskotekastar, 2023-03-11:
**Is your feature request related to a problem? Please describe.**
---

**Describe the solution you'd like**
The ability to log in with an account on multiple devices and thus sync all slicer settings.

**Describe how it would work**
The user would be able to register and then log in.
He would log in to slicer on, for example, a computer and a laptop and could choose the option "synchronize settings" and this would synchronize his slicer settings, printers, filaments, ... .

**Describe alternatives you've considered**
I've always had to backup the configuration on one device and restore the backup on the other.

**Additional context**
---

jDavidnet, 2023-06-20:
I am not a commercial user or printer (yet), but I do have PrusaSlicer installed on my desktop, my laptop, and my wife's laptop, and I would love to have most of the settings synchronized between all of our devices.

For printers, filaments, hotends, etc... I'd like to be able to check a box on the saved profile and be able to synchronize it with my group/family.

Right now, she is likely to use my profiles, but it should be an option for people in the group to have private or personal settings in addition to the synchronized ones or the synchronized root profiles.

Maybe a GitHub repo can either be used in the interim or with such a sync feature.  I'd be willing to pay a few dollars a year for a settings sync feature and more per year to sync another family/group files.

alexiri, 2023-07-29:
This seems like a duplicate of #4898.