prusa3d/PrusaSlicer #10027: Changes to Cut tool:  Snap to 45º and 90º and make Z heights persistent

silvermanphoto, 2023-03-11:
1.  It is difficult to create a 90 cut because the angle of the cut tool is variable without snapping to common angles like 90º or 45º, and there is no way to enter an angle manually.  Could you please create a text entry for the cut angle, and snap to 90º or 45º changes?

2.  Often when you make cuts of multiple items it is helpful to make the same cut repeatedly.  But, when you select the cut feature, a default Z height is used.  If the user enters, say 6mm for a cut, the next time you select the cut tool, 6mm should still be the entry, at least within one PrusaSlicer session.

YuSanka, 2023-03-13:
Thanks for your report.
>It is difficult to create a 90 cut because the angle of the cut tool is variable without snapping to common angles like 90º or 45º, and there is no way to enter an angle manually. Could you please create a text entry for the cut angle, and snap to 90º or 45º changes?

There is 5 and 45 degrees snapping for a rotation of the cut plane. It works in same way as for object rotation.
[image]

> Often when you make cuts of multiple items it is helpful to make the same cut repeatedly. But, when you select the cut feature, a default Z height is used. If the user enters, say 6mm for a cut, the [...]

I don't understand your workflow. 

If you need to cut the same object several times with a same cut settings, than you can to increase a number of object instances. And cut will be performed to all instances at a time.

But if there are different objects with different sizes on a print bed, then cut setting will be reset for new selected object every time. And it's not a bug, it's requirements.

DutchessNicole, 2024-06-21:
I'm having the same issue. When manually defining a cut it is not possible to set the cut parallel or perpendicular to a plane. So if I want to cut a model straight up, and I'm off by a few degrees, there's no way to accurately change the cut to be precisely vertical. The 45 degree snaps seem to be based off of the initial angle, so if I was 2 degrees off on my initial cut, then all 45 degree increments will also be 2 degrees off. A way to specify the exact angle as a text input would be great.

davbow123, 2024-09-12:
I would like to see the rotation/cut compass have an option to stay aligned with the x,y,z axes.  Like @DutchessNicole indicated, once you are off, it is difficult to get reoriented.

closed 2026-09-29 by prusaslicer-bot[bot]