prusa3d/PrusaSlicer #10029: Access Violation

rafaelldavid, 2023-03-12:
### Description of the bug

[image]

I'm trying to slice a very large piece, so I decided to cut it in half through the program itself.
After separating the parts, when ordering to slice, this error message appears.
If you use version 2.5 for slicing, it works normally.

### Project file & How to reproduce

I uploaded the files to my google drive as they are large.
The file itself comes with some errors that can be fixed via netfabb in the program itself.
The whole piece slices normally, but if I cut it in 2 and place the flat parts on the table to slice, the error occurs.
https://drive.google.com/file/d/1m5_W7oseH728alN8f74XwitcuVGGp11g/view?usp=share_link

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Versão 2.6.0-alpha5+win64

### Operating system

Windows 11 21H2

### Printer model

Ender 3 with Klipper

rafaelldavid, 2023-03-12:
Performing some tests, I could also notice that, if I slice the piece without changing its orientation, it slices normally.
The problem happens when I place the flat face on the print bed.

rafaelldavid, 2023-03-14:
Buddy, I tried it again today and it worked. I reproduced all the same steps as before, but the error didn't happen again.
I can't say now if it was a problem related to my computer, the part I'm using or something in the slicer/cutter of the part.
I'm sorry, if it happens again, I'll send it here...

kubispe1, 2023-03-14:
Could be older alpha...If you will find trouble let me know. 
For now, I am closing the issue.

closed 2023-03-14 by kubispe1