prusa3d/PrusaSlicer #10030: Complete Individual Objects As a Height Range Modifier

RickR74, 2023-03-12:
If its possible, I think it would be really useful to have a "Height Range Modifier" that allows you to introduce an output option to complete individual objects, at a certain height in the print rather, than only though out the entire model printing time.
Why? Because I have a lot of reasonably large models I print that start off with a large printing area, followed by various small or thin objects, that are on top of the larger model surface area below.
In many cases the small objects are spaced well apart and would be suitable for individual printing.
The obvious reason for wanting this ability is to prevent excessive stringing caused by the fact that the printing speed, etc, has been slowed right down for the small objects to be printed. Reducing temperature, flow rate and other settings through the use of the existing height range modifier settings helps, but isnt a complete solution for stringing issues. The problem with stringing at this level of the print is also increased due to the strings dropping down onto the finished larger surface area, damaging the surface area.

I couldnt find any way to make this happen, but if there is a way, please ignore the above!
And yes I have spent a lot of time trying to prevent stringing on very small objects, but with little success.
Am definitely open to suggestions on improving print quality on small objects, by using modifiers.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe how it would work**
Be as descriptive as your skill level allows you.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or paint drawing about the feature request here.

RickR74, 2023-03-19:
Well I guess no interest in this subject, which I suspected due to the fact that most small objects on a larger bottom area wouldnt have sufficient clearance from each other and the hot end. So that would make it a fairly rare scenario where a mod like that could be of use. Just have to find another way!

closed 2023-03-19 by RickR74

ulab, 2023-03-19:
Things don't happen in a week…