prusa3d/PrusaSlicer #10031: Add a setting to enable Z-Hop for travel moves between parts only.

JagguRaja, 2023-03-12:
**Is your feature request related to a problem? Please describe.**

It is somewhat linked to a problem but that problem has a solution. The issue is, it is not a very good solution. The problem is, when printing multiple parts and the top surface is flat, sometimes when the nozzle moves over to the next part it travels through the top surface and digs into the part a bit leaving behind a travel mark/line. 

The obvious solution is to enable z-hop, however, this solution adds unnecessary time as you don't need to have z-hop enabled for EVERY retraction.

**Describe the solution you'd like**

If there was a setting like "enable z-hop for travel moves", it would be the best of both worlds I believe. The print time would remain relatively unchanged and travel moves would not leave nozzle marks on the top surface.

It would work by identifying travel moves between parts (or just travel moves in general as this can be useful on single part multi surface prints) and adding a z-hop right before all those moves.

**Describe how it would work**

Maybe it can be a post processing script. It looks for G1 commands that do not have an associated "E" (or extrude) command and adds a z-hop before those moves.

I am sure people smarter than I am can come up with a more efficient approach.

**Describe alternatives you've considered**

Of course there is the solution I mentioned in the first section but like I said, that adds extra time that does not need to be added. There is also the solution of flow but in my case (and a few others I have spoke to) have calibrated flow as best we can based on Prusa's flow calibration guide but the travel marks still remain.

This new solution will in my opinion give the best of both worlds for z-hop where it applies. It will remove travel marks and add next to no extra time unlike the current all or nothing solution the slicer currently has of either enabling or disabling z-hop.

**Additional context**
N/A

MrBlackmidi, 2023-03-27:
an issue of mine that is somewhat related to this and others that are not #10066 #9285 #9885

Boerny41, 2025-01-28:
Would be great

dHutchings, 2025-07-12:
Seconded, would be great.

closed 2026-09-29 by prusaslicer-bot[bot]