prusa3d/PrusaSlicer #10032: Support generated through objects

federico9292, 2023-03-12:
### Description of the bug

Organic supports are generated through the object ignoring perimeters, infills, etc. 

it can be bypassed incrementing the distance between models in auto-arrangement but for a specific distance, if the base of the organic support isn't enough it just ignores the problem and puts the support through the object 

### Project file & How to reproduce

[image]

[organic supports bug.zip](https://github.com/prusa3d/PrusaSlicer/files/10950373/organic.supports.bug.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-alpha5

### Operating system

Windows 10

### Printer model

Ender 3

GordonW-Printer, 2023-03-18:
I have observed similar problem with default auto-arrange distance (6), but increasing to (10.75) seems to work with my particular objects.
The images are from the bottom at 6 and 10.75 to demonstrate issue.

[image]
[image]

Tiri1i1, 2023-03-20:
I have seen the same issue on the Alpha 5 with default arrange spacing of 6mm
[image]

foreachthing, 2023-03-21:
https://github.com/prusa3d/PrusaSlicer/issues/316

kubispe1, 2023-03-23:
Due to the enormous amount of issues, we are discussing how to handle the collisions. 
Thanks for understanding
I am closing the issue. It is a duplicate.

closed 2023-03-23 by kubispe1