prusa3d/PrusaSlicer #10034: PrusaSlicer 2.6 alpha5 stuck

SiberSilverFox, 2023-03-12:
### Description of the bug

When attempting to slice a file at .2 the process gets stuck at 70% and does not go any further
If i slice at .25 draft then it works

[image]

[image]

[image]

### Project file & How to reproduce

[Hex Platform.zip](https://github.com/prusa3d/PrusaSlicer/files/10951335/Hex.Platform.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.6.0-alpha5+win64

### Operating system

Windows 10 (build 19045), 64-bit edition

### Printer model

PrusaMini

3Dtcb, 2023-03-12:
I have the same problem.
I tried to slice this [file](https://www.thingiverse.com/thing:5901805)  and it stops at 70%.

It doesnt matter if I use a profile based on 0.2 or an original profile from Prusa. 

PrusaSlicer
Version:   2.6.0-alpha5+MacOS-arm64
Build:     PrusaSlicer-2.6.0-alpha5+MacOS-arm64-202303061441

Operating System:    Macintosh
System Architecture: 64 bit
System Version:      macOS Ventura Version 13.2.1 (Build 22D68)

Printer model
Elegoo Neptune 3 Plus

C75mitch, 2023-03-12:
I have the same issue with a mammoth head I was trying to slice. Works fine in 2.5.0, but not in version: 2.6.0-alpha5 Again, stuck at 70% and I left it for 3 hours. Tried multiple times with different settings but same result.

Windows 10 64bit

Version: 2.6.0-alpha5

Prusa i3 MK3S

alltjagvill, 2023-03-16:
I have the same problem, but only with organic support and after I changed Extrusion width for support material to 0.18.
Gets stuck at 70% Generating supports.
Setting it to default value or 0 works no problem.

Windows 10 64but
Version: 2.6.0-alpha5
Ender 3

3Dtcb, 2023-03-21:
I encountered the problem again and when I switch to edit mode and back to preview mode then it looks as if Prusa cant fill the object.

https://imgur.com/a/ZIywUQ3

I sliced the file in Cura and it was working without a problem. I assume it has nothing to do with the file itself.

kubispe1, 2023-03-23:
Thank you guyes, from the first look it is cause by arrachne end with: 
- PrusaSlicer-2.6.0-alpha5+172
`Slic3r::Arachne::SkeletalTrapezoidation::separatePointyQuadEndNodes()`

But let's see after more deep investigation.

- SPE-1608

lukasmatena, 2023-04-02:
Fixed in [2.6.0-alpha6](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha6). Closing.

closed 2023-04-02 by lukasmatena