prusa3d/PrusaSlicer #10035: 2.6.0 Alpha 5 embossed text z height

wildwick69, 2023-03-12:
### Description of the bug

when embossing a text to a loaded stl then the sliced object doesn't show up the correct z-height,
eg overall Z height of the STL is 2.6mm and add an embossing text with 1.4mm the resulting overall z height is 3.3mm...
when doing a colour change @ 2.8 it only shows 3 additional layers instead of 6 (0.2mm layer height).
the text also impacts lower layers from starting full top layers where it shouldn't do so

### Project file & How to reproduce

[pickguard-offset-mike-magneto.zip](https://github.com/prusa3d/PrusaSlicer/files/10952211/pickguard-offset-mike-magneto.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0-alpha-5

### Operating system

Windows 10

### Printer model

Prusa MK3S+

Jony01, 2023-03-13:
Sorry for bad talking to user by GUI. It is only missunderstanding.
When you don't use surface of model, than the text volume is a half behind the point on the surface and a half infront.

Text could be placed on the coarse(not flat) surfaces where it will create gap between model and letters.
Thanks for feedback. We will discuss it more deeply internaly.

wildwick69, 2023-03-13:
hmm but I used the surface, as the beginning of the text is @ Z 2.6 in slicer..   increasing Z for lifting up the the text (assuming half above/half below surface) gives me warning that theres a gap between surface and text. Also the height in the gcode (output of octoprint) tells me a overall height of 4.3mm with printing 3.2mm in reality (and as slicer is showing)

closed 2026-09-29 by prusaslicer-bot[bot]