prusa3d/PrusaSlicer #10038: Automated organic support doesn't touch de desired object

GabrielMossi, 2023-03-13:
### Description of the bug

### Organic support get close to the overhang but didint tuch, boddon supports works just fine. 
[image]
[image]

### Project file & How to reproduce

[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.6.0-alpha4+win64-202302180012

### Operating system

Win10

### Printer model

Ender 3

jmd73, 2023-03-13:
@GabrielMossi You should use 0 interface layers.

From the release notes:
Interface layers are not always produced reliably. As of now printing with zero interface layers is recommended, which produces easy to remove supports.

So set this setting to 0: (off).

[image]

GabrielMossi, 2023-03-17:
Hello, @Jan-Soustruznik, I tested the same file on alpha5 release, same configurations.
the problem persists. 
 
[image]

Jan-Soustruznik, 2023-03-20:
Hello, @GabrielMossi , could you share .3mf file, please? Thank you

Jan-Soustruznik, 2023-09-18:
Issue closed

closed 2023-09-18 by Jan-Soustruznik