prusa3d/PrusaSlicer #10039: Add the ability to set separate (Print, Filament, and Printer) settings for each object in sequential printing.

ccdanieldb, 2023-03-13:
Currently i am unable to change Max_volumetric_Speed setting on indivual objects during sequential printing slicing.  I would like to be able to do so.  

Can you add the ability to set separate (Print, Filament, and Printer) settings for each object in sequential printing.  This would take care of the issue

When sequential printing is active I would like to be able to select each object on the right side of the screen and assign separate (Print, Filament, and Printer) settings.  

I can see that something like this is already there and that by right-clicking on the individual objects I can (Add Setting) but not all settings are there.  

If what I am asking for is not possible can you add (Print_Settings/Speed/Audo_Speed_(advanced)/Max_volumetric_Speed:) to the current (Add settings) method?  Thank you for your consideration.

Bumblebiber, 2023-03-13:
It would be great if there were more individual settings in general. Like retraction settings are completely missing.

ccdanieldb, 2023-03-14:
Bumblebiber,  Thank you for posting.  You are correct one option is to have all print and filament settings accessible in the add setting section.  But if this can't happen for some reasion, being able to set separate (Print, Filament, and Printer) settings for each object in sequential printing is an other option.

closed 2026-09-29 by prusaslicer-bot[bot]