prusa3d/PrusaSlicer #10042: Retraction is still done in vase mode

jsrobson10, 2023-03-13:
### Description of the bug

It makes sense why there'd be retraction code at the starts and ends of a vase mode print, but when geometry is slightly more complex it'll still generate retraction and unretraction steps. These should not be here. 

A workaround that I have found works is just setting retraction to 0 and disabling firmware retraction. This'll completely emit retractions/unretractions from the gcode. 

### Project file & How to reproduce

# my project file
[water_bottle.zip](https://github.com/prusa3d/PrusaSlicer/files/10957249/water_bottle.zip)

# the issue
[image]

# steps to reproduce
1. import an STL that is slightly more complex, eg a cylinder with extra bits, such as one like I've attached.
2. enable spiral vase mode.
3. have software retraction as a >0 number OR have firmware retraction enabled.
4. slice and export.
5. inspect the gode, searching for G10/G11/negative E moves where they should't be.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0

### Operating system

Arch Linux

### Printer model

Creality Ender 3 v2 (Running Marlin 2.1.2)

jsrobson10, 2023-03-16:
also gonna add, whilst the workaround does definitely improve things, there is still layer change gcode left over which results in a slight pause in marlin since there's just completely unnecessary stuff there. it would be needed if the next "layer" was on a different print speed as the last, but since it's not, it's not needed. also that `G92 E0` is completely unnecessary since relative extrusion is enabled. it just slows down execution leading to pauses.

sensaurabh, 2023-03-28:
Seeing the same thing... Retractions on layer change in vase mode on a simple cylinder primitive. Setting the retractions to zero causes 'G1 E0 F0' instructions in the G-code stopping print.

schuetzi99, 2023-06-16:
I have the same problem with PrusaSlicer 2.6.0-RC2

imadcat, 2024-06-23:
Has this problem been addressed?

MasterIO02, 2025-08-27:
More than a year passed and this still hasn't been fixed. Vase mode prints are impossible, as disabling retractation as well as wiping still produces useless g-code instructions that slows down the print leading to artifacts.

ars-java, 2026-09-28:
I have the same problem. It must be fixed.