prusa3d/PrusaSlicer #10044: PrusaSlicer Does Not Print Perimeters in Same Direction (Support for Dual Color Filament)

cyon1c, 2023-03-13:
**Is your feature request related to a problem? Please describe.**
When trying to print an object in PrusaSlicer 2.6.0-alpha4, perimeters are not always printed in the same direction, resulting in an odd surface finish when using dual color filaments like Matterhackers Quantum PLA.

I've attached a .3mf file of a print I just did, as well as a photo of the print to demonstrate the issue..

[image]

[Bounty Badge Case.zip](https://github.com/prusa3d/PrusaSlicer/files/10958282/Bounty.Badge.Case.zip)

r3Fuze, 2023-03-13:
Related feature request: https://github.com/prusa3d/PrusaSlicer/issues/7575

Jan-Soustruznik, 2025-04-10:
closed as duplicate of #7575

closed 2025-04-10 by Jan-Soustruznik