prusa3d/PrusaSlicer #10045: Model disappear when hollowing

trev0r-STA, 2023-03-13:
### Description of the bug

When I try to hollow a model, it disappear, so it's very hard to put drain holes :-)
I can put holes (see screenshot)
I've tried with differents models, same thing.
If you need more information, ask me
Thanks for reading

### Project file & How to reproduce

[Venus.zip](https://github.com/prusa3d/PrusaSlicer/files/10959432/Venus.zip)
[image]
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.6.0-alpha5+linux-x64-GTK3

### Operating system

Debian 11.6

### Printer model

Elegoo Mars 2

kubispe1, 2023-03-15:
Hi, thanks. We will check it and try to improve it.
SPE-1449

trev0r-STA, 2023-04-05:
Hello
Just test with the alpha6: same thing. But find a workaround: if I set Clipping on view to 0.01, the model appear and I can set holes where I need.

Hope this help

FidelCapo, 2023-04-14:
This issue should be fixed and fix will be part of next release.
Closing.

closed 2023-04-14 by FidelCapo