prusa3d/PrusaSlicer #10046: large object causing thermal runaway alarm

LHooven, 2023-03-13:
### Description of the bug

Unsure if this was addressed or even known.  I have a large (over 1M triangle) print i was trying to do, 4 days worth actually.  Ender v2 neo printer, and i think the size of the print is causing thermal runaway (according to the board) i sat and watched it jump from 190 to 205 back and forth, and it would then trigger thermal runaway.  grabbed another file from CURA (dont hate me they are just tools) and the temp has been rock solid, i re-sliced the item in question in cura and will see if it does the same thing.  I'll also download alpha 5 and see what's new 😀

i cant attach the file as its too large.

### Project file & How to reproduce

[Deserts_Kiss_-_Diorama_Dice_Tower_-_v1.2.5.zip](https://github.com/prusa3d/PrusaSlicer/files/10961857/Deserts_Kiss_-_Diorama_Dice_Tower_-_v1.2.5.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0 alpha4

### Operating system

win 10

### Printer model

creality ender v2 or 3 neo

MrBlackmidi, 2023-03-14:
Afaik, thermal runaway is from the printer, not the slicer and i doubt any gcode exists that triggers thermal runaway

Gunner087, 2023-03-18:
I sliced the project file using 2.6.0 alpha4 and searched the produced gcode file for M104 and M109 commands. None were present except at the beginning and end of the file. I think this is a printer issue and not the slicer. Just going to guess that it might be related to different cooling fan speeds between the two files that were printed or just an intermittent bad electrical connection from a wire that is beginning to fatigue.

MrBlackmidi, 2023-03-20:
agreed

friendlyFriend4000, 2023-04-09:
On my mk3s,  I have the same problem with prusa slicer  2.6.0 alpha4 and alpha6 . This is issue does not occur with Prusa slicer 2.5.0 or 2.5.2 or cura. It is to note that the thermal runaway( temp too low) occurs usually around the same time, around 30 - 40 minutes after starting the print . Here are two failed prints from today. My third and current print does print without any thermal issue
[image]
[image]

Porsche911Dakar, 2024-01-13:
I also have the same issue with my ender 3 v2 Neo even replaced my thermistor and issue still persisting. Seems to be for me around the time when second layer temp shift it just doesn't like it and gives me the same error

LHooven, 2024-01-14:
on my neo i found the tiny screw that holds the heater in the block had worked loose its at the front in the center, and is the smallest allen wrench that comes in the kit. try tightening that a tiny bit, i did that and have not had an issue since.

Porsche911Dakar, 2024-01-14:
Thanks!, Tried that and seems to be fixed also found my fan was very much loose aswell. lol

Porsche911Dakar, 2024-01-16:
That didn't last long, lasted till yesterday got one print done inbetween fixing and another failed print. **BUT** I have new info it seems to do so when the part cooling fan is enabled the temperatures drop around 1 degree per second then you get the normal error 'nozzle temperature too low' Anything to help would me greatly Appreciated 😁

Porsche911Dakar, 2024-01-21:
it seems to work perfectly with old sliced files its worked every time with old pre-sliced files and doesn't on current Prusa 2.7.1. @josefprusa please fix 🙏🏻

closed 2026-09-29 by prusaslicer-bot[bot]