prusa3d/PrusaSlicer #10048: Organic support generation hangs the slicer at 100% CPU

kozross, 2023-03-13:
### Description of the bug

When trying to slice a model using a 0.2mm nozzle profile I designed, organic support generation hangs PrusaSlicer, while using 100% CPU. I tried to let it run for about 5 minutes, but even after that, it still wasn't done. This doesn't happen on any other kind of support (Grid or Snug) as far as I can tell, but it seems to happen regardless of which model I try.

### Project file & How to reproduce

[Project zip](https://github.com/prusa3d/PrusaSlicer/files/10962862/slicing-crash.zip).

How to reproduce:

1. Open the project I provided
2. Try to slice
3. Watch the CPU burn bright while nothing happens and no progress is made

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.6.0-alpha5+linux-x64-GTK3

### Operating system

Arch Linux

### Printer model

Prusa MINI+

lukasmatena, 2023-03-14:
Thanks, looking into it.

lukasmatena, 2023-03-14:
Fixed for 2.6.0-alpha6 (SPE-1595). Thanks again for reporting it.

Queso2469, 2023-03-25:
I noticed this happened to me once when I tried to increase Tip Diameter setting from 1 to 1.5. Worked again at 1.25.

kubispe1, 2023-03-29:
This issue is tested in 2.6.0-alpha5+245. The fix will be available in 2.6.0-alpha6.
Thanks and have a nice day.