prusa3d/PrusaSlicer #10050: breaking the perimeter with a modifier

uklugi, 2023-03-14:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
[image]
**Describe the solution you'd like**
A clear and concise description of what you want to happen.
it is necessary that the number of parameters increases only when the modifier is found along the perimeter
[image]

**Describe how it would work**
Be as descriptive as your skill level allows you.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or paint drawing about the feature request here.
in cura, the model remains intact

neophyl, 2023-03-14:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/7762 (and several others.  This has been reported many times already.

foreachthing, 2023-03-14:
> in cura, the model remains intact

Wha....! OK!! 😮 
Back here: https://github.com/prusa3d/PrusaSlicer/issues/6881, Cura also broke the perimeter. So ... no more excuses, PS-Devs?!?!

closed 2026-09-29 by prusaslicer-bot[bot]