prusa3d/PrusaSlicer #10051: Supports deleted when reload from disk

DonBorste, 2023-03-14:
### Description of the bug

If you using supports by painting it to the necessary areas (enforce) and reload the original file from disk (caused in minimal changes), the painted supports are deleted completely. 

### Project file & How to reproduce

1. load a part wich needs supports in some ore one area
2. enforce supports in this areas by painting it
3. slice it
4. change the part design lightly in another application and save it
5. reload the part from disk by right klick the part on sheet plate and select "reload from disk"
6. slice it
7. the support are gone :-(

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0-alpha5+win64

### Operating system

Windows 11

### Printer model

Prusa i3 MK3S

neophyl, 2023-03-14:
Thats because the painting is reset when a model is reloaded or replaced.  The painting requires the list of surfaces of the model (done is some form of list).  When reloaded or replaced the list is reset as its different for a different model.  
And yeah you could reload the same model and it also goes but then why are you reloading the same model ?  From a software code point of view it has to be reset to work properly.

kubispe1, 2023-03-15:
Same for other painting features.

revaip, 2025-03-13:
The real problem is: the paint-on feature is not dropped, so the user can assume that everything is good, but from now it's an empty feature. Perhaps it would be nice to the slicer to notify the user that the paint-on features need to be redefined, moreover, the empty features sould be deleted from the parts/feature tree from the concerned instances (obviously it should stay there at the copied instances, since they're not reloaded).

https://github.com/user-attachments/assets/2aa35269-66c0-49a5-8f3b-8134af12a57a

closed 2026-09-29 by prusaslicer-bot[bot]