prusa3d/PrusaSlicer #10055: Generation of lightning infill fails "Infill failed"

KayGestalten, 2023-03-14:
### Description of the bug

Slicing of the model fails after some time, with the message "infill failed", when using the lightning infill.

Currently I detected the issue only using this model
[image]

there it works

### Project file & How to reproduce

[jinx-base.zip](https://github.com/prusa3d/PrusaSlicer/files/10969199/jinx-base.zip)
Attached the project 3mf

Slicing using the 2.6.0 alpha5 produces the issue

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0 alpha5

### Operating system

Windows 10

### Printer model

Prusa MK3S+

adamgreb, 2023-03-15:
I'm seeing the same thing with the alpha5 release that I do not see with 2.5.0.

kubispe1, 2023-03-15:
Hi, we will look into it. Thanks
SPE-1578

Ockenfuss, 2023-03-15:
Here is another example with the same issue. For me, slicing works with lightning, but if I add a height-range modifier with grid infill at the bottom, slicing fails. The previous version, 2.6.0 alpha4, seems to work fine. Setup: Windows 11, MK3S. See the attached example.

[MountainRelief_slicerError.zip](https://github.com/prusa3d/PrusaSlicer/files/10985294/MountainRelief_slicerError.zip)

spasecookee, 2023-03-20:
I have provided a bit more detail with my bug report here: https://github.com/prusa3d/PrusaSlicer/issues/10115

kubispe1, 2023-03-29:
We tested these issues in **PrusaSlicer-2.6.0-alpha5+266** and fix will be part of the **PrusaSlicer-2.6.0-alpha6**

[image]
@Ockenfuss , this is probably another issue, but it is recently also fixed. The model is quite heavy, please check it after release 2.6.0-alpha6 and let us know.

[image]

I am closing it. Thanks

closed 2023-03-29 by kubispe1

Ockenfuss, 2023-04-01:
Thanks, the new version alpha6 seems to solve the issue! Nevertheless, there is a new bug when applying lightning infill on top of another infill. I opened an issue here: #10222

kubispe1, 2023-04-05:
@Ockenfuss  Thanks