prusa3d/PrusaSlicer #10056: Hole locations go wrong when doing a split and adding dowels (aka "holes both sides")

NKT00, 2023-03-14:
### Description of the bug

I couldn't find an open issue for this one.

I have a U shaped part with round cross section, and when I split it in half, in the slicer, everything works. If I add pins to it, that also works. But, if I ask for dowels, it doesn't do the combine before moving the split part, so the holes aren't generated in the right place.

I'll post an image later, and dig into it a bit more. 

### Project file & How to reproduce

To-do

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0 alpha5 

### Operating system

Windows 10

### Printer model

Prusa 3S

NKT00, 2023-03-14:
Steps to reproduce:
Import the STL from [Assem1 - shackel2-1.zip](https://github.com/prusa3d/PrusaSlicer/files/10972327/Assem1.-.shackel2-1.zip)

1. Turn it 90 degrees so it is "flat" and parallel to the bed.
2. Initiate a split. It will by default want to cut it neatly in half. Leave it like that.
3. Select "Edit connectors". Don't change the default settings.
4. Add 1 or more "holes" by clicking on the model.
5. Click "Confirm connectors"
6. Perform the cut.
7. Note that the "holes" are in the wrong place, and not on the model.

neophyl, 2023-03-14:
I think this one is related https://github.com/prusa3d/PrusaSlicer/issues/9991

NKT00, 2023-03-14:
Yes, very closely! I'll look into it on the laptop.

kubispe1, 2023-03-15:
The issue should be fixed in the next release alpha6.
I am closing this issue as a duplicate of #9991

closed 2023-03-15 by kubispe1